x-station/xstation-issues #393: Inconsistent booting after working for about a year

VanFinale, 2024-12-29:
So I installed the x-station myself about a year ago and it was working fine for a while. Occasionally it might fail to boot on initial start-up but a reset would normally fix it. Now it getting harder and harder to boot. Will hang for a while if the cover is closed before going to the memcard screen. Blue LED is on green will flash for a second on startup. It's installed on a PU-18 and I opened it back-up and re-seated the flex cable just in case. I attached some pictures of my installation I took when I re-seated the cable. I've tried multiple SD cards and re-formatted them all with the SD Formatter tool as well.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2024-12-29:
Hey,
I'd like you to focus some cleanup work on the 701 IC (the DSP). The lifted pins should be straightened out a bit, to ensure they never short.
The pads below those lifted pins can be reflowed with your soldering iron, by just touching them once with the hot iron.

Ultimately, I think the briefly flashing blue LED is a hint: That is the power LED, and it should only ever come on and stay on perfectly the same.

VanFinale, 2024-12-29:
The green LED briefly flashes on boot. The blue LED stays on consistently.

> Hey, I'd like you to focus some cleanup work on the 701 IC (the DSP). The lifted pins should be straightened out a bit, to ensure they never short. The pads below those lifted pins can be reflowed [...]

VanFinale, 2026-09-29:
I reflowed the pads and moved the lifted pins further apart to ensure they won't touch. It started working again for a day but now it is back to not booting.

[image]
[image]
[image]

ramapcsx2, 2026-09-30:
Please test for shorts here:

[image]

Intermittent posting could have to do with the Lid Switch as well. Sometimes the shell does not properly close that contact.
You can try leaving the lid open, and jamming a Q-Tip in the sensor instead.

VanFinale, 2026-09-30:
I tested it, there are no shorts between those 2 pads. I checked the switch as well and it still fails to boot while holding it down manually. With the top on if the lid is open it goes to the memory card menu right away. When the  lid is closed it holds on the white boot screen for an extra 5-10 seconds before eventually going to the memory card menu. So the lid being open or closed does make a difference so I am pretty sure it is working.

[image]

ramapcsx2, 2026-10-01:
Yes, the switch should be working then. So I'm out of suggestions at this point. Intermittent booting is almost guaranteed to be one or more weak solder joints, likely on the QSB somewhere. You can measure them, and by the pressure of the probes, you see good contact, when there might not be. Perhaps look for a local modding service?

VanFinale, 2026-10-11:
I realised there is a test point diagram in the troubleshooting section. Found a few of the pins had inconsistent connections. It seems to be booting every time now. I'll keep the issue opened a bit longer just to make sure it stays that way. Thank you for all your help.

closed 2026-10-18 by VanFinale