x-station/xstation-issues #394: [Feature Request] Auto playlist generation when launching multi-disc games.

me0262, 2025-01-01:
Title says it all.

Test procedure:
1. Put title with multiple discs in folder, suffixed with a -disc# tag (eg. FF7-disc1.iso, FF7-disc2.iso, FF7-disc3.iso)
2. xStation selector should list title by first disc only (FF7-disc1.iso or FF7 (multidisc).
3. Launching multidisc title should auto create playlist of all NAME-discX.iso files.

How I would implement (but it's not up to me):
Seek to end of string (or last '.') and see if reverse string matches osi.*csid (careful not to go out of bounds on games like N2O or ODT).

Hopefully there should be enough flash memory to implement this.

ramapcsx2, 2025-01-01:
Hey,
this is a common request, and I'll eventually get it done.
There's a few ideas about the implementation, but it won't be via some kind of string detection, looking for "multi" or such schemes.
2 basic problems are that sometimes, multidisk games come organized in a folder per disk, sometimes all disks are in the same folder.
So I'll probably look for cue files that indicate that this will be multidisk, plus one extra check into any subfolders. If then this all seems like a queue set, I can automate it.

b0h, 2025-08-09:
Hi, it's been a while. Any news regarding this feature?
it's sad to see no new firmware revisions lately.

ramapcsx2, 2025-08-10:
I'm sorry, nothing yet, and it will be some time more. I'm currently working on the new mStation project, and it's taking up all my time really.
I put a lot of work into the 2.2.0 beta last year, to prepare for this, so if you don't have that public beta yet, it's on Discord.

Again, sorry, but I can only really do one big thing at a time ><