x-station/xstation-issues #395: Xstation not working properly

DOK219, 2025-01-24:
I installed my Xstation onto a 5501 console. Worked twice and now no display or sound when powering on. I have tested all the continuity, rechecked the QSB and resoldered the APL wire. Nothing works. Xstation powers on when connected and the green light flashes twice for the SD card. Any thought on what this could be? Thank you in advance.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

ramapcsx2, 2025-01-24:
Please ensure that the APLL wire doesn't bridge into the wrong signal here:

[image]
as it looks awfully close. (a thinner wire should be used)

[image]

The fault being that there is no picture, I assume that either the APLL signal doesn't arrive at the PSX CPU properly, or that at some point, one of the PSX board fuses blew.
So you also want to check the PSX fuses for continuity, and if one is blown, then there was/is still a short somewhere in the system that needs to be fixed (and then that fuse replaced)

ramapcsx2, 2025-01-24:
The fuses on PU-18 are the small black parts around the power connector that say "PS605" etc next to them.

[image]