x-station/xstation-issues #396: Boot sometimes hangs on white Sony screen

yayitsd4n, 2025-01-27:
This might not be an xstation issue, but I wasn't sure where was best to ask. Every once in a while (maybe once in fifteen or twenty times), when I turn the system on, it will hang on the white Sony screen, never moving past it to the xstation UI or the the memory card screen.

The only other mod I have installed is a new picoPSU with USB C power delivery. I did manage to install the xstation myself so I definitely could have done something wrong, but I'm not really sure where to start to diagnose this. Otherwise the system works perfectly, which makes me think that issue might not be unrelated to the xstation or the install.

ramapcsx2, 2025-01-28:
Hm, there's not much I can think of that would give this behaviour. 
If the I2S word clock was intermittent (signal 15), it can do this, but then you would see many more problems with games (not) running..

[image]

yayitsd4n, 2025-01-28:
Thanks for the response. I haven’t come across someone yet with my exact issue, so I’m not surprised you can’t think of much that might cause it.

I’ll take the shell off, inspect everything, and put the old psu back in. I don’t recall having this issue until after I put in the picoPSU, but it’s very possible that I did since I haven’t put many hours on the console yet. I might as well eliminate that possibility since it takes next to no effort to change.