x-station/xstation-issues #398: Install looks fine but only gets Main Menu screen with blue LED

tiago-sch, 2025-02-13:
Did the installation following the guides, even did a check with the test points and everything seems fine. Did this installation twice in two consoles and have the exact same issue in both.

The only different thing both have is a "jump" for loading different region games (very common here in Brazil, almost impossible to find a console without one of these). Could this be the culprit? Or am I f***ing something in the install?

[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2025-02-13:
Hey, this here looks bad, but not sure it would cause the problem:

[image]

In general, the board looks quite worn. There's work here on the old drive connector that looks like it might short:

[image]

The video encoder thing is unrelated.

tiago-sch, 2025-02-13:
Tks for the quick reply!

I cleaned up the connector but I believe it was not shorting was just high for some reason. The first short I was able to cleanup as well but I noticed that the console is kinda unstable with power on, if I move it a bit or open the lid the console shuts down (including the green led) so Im thinking its a bad board (two, actually) :/

Thanks anyway!

ramapcsx2, 2025-02-15:
Not sure how to help further, but you need to solve the shorts, repair any open fuses and such..