x-station/xstation-issues #399: Late PU8 install problem

spikeyxx, 2025-02-21:
Hi, looking for help.

I'm working on an SCPH 1002 Late PU8 board, following the guide.

I lifted the pins, removed the resistor and filter, then flipped and removed the 3 resistors on the back.

I then installed the XStation, did a test boot and had the following problem:

System will not boot. There is an audible whine coming from somewhere.  The green light does not turn green.

When I remove the main board from the PSU and turn it on, the wine isn't present and the green light comes on.

I tried removing the XStation and retested, however the same problem persists.

Is this a well known problem?  Does it happen with incorrect installation and would installing correctly fix (i.e. due to the lifted pins and removed components) or does it sound like something else?

Thanks

ramapcsx2, 2025-02-21:
Hey,
the xStation doesn't get power from the mainboard. Points responsible are "2" for +8V and "1,11,18" for ground:

[image]

ramapcsx2, 2025-02-21:
Sorry, the ground point was just out of frame. It's the connection to the capacitor here:

[image]

spikeyxx, 2025-02-22:
By main board, I mean the playstation motherboard.

spikeyxx, 2025-02-22:
What I'm wondering is, does the process of lifting the pins and removing the filter and resistors create an "unbootable" playstation, that would exhibit the whine and failure for the LED to go green - is that expected behaviour.

If so, would a successful installation remedy the problem OR does this sound like something else entirely - i.e. the installation fried nearby components or created shorts?

I'm unsure how to proceed due to my lack of knowledge on the matter.

ramapcsx2, 2025-02-22:
Once the parts around the oscillator for the CPU are removed, the console will need the xStation connected to start. It supplies the clock then.
The xStation has 2 LED: blue for power, and green for SD card access. You would want to at least see a permanent blue LED on the xStation, indicating it has power.
Since that's missing, it follows that the 8V / Gnd towards xStation are missing.

spikeyxx, 2025-02-22:
> Once the parts around the oscillator for the CPU are removed, the console will need the xStation connected to start. It supplies the clock then.
> The xStation has 2 LED: blue for power, and green for [...]

Would this cause the whine and the lack of green light on the playstation power switch?

When I had the XStation installed, it was briefly showing blue, then going off.

ramapcsx2, 2025-02-23:
If your PSX power supply isn't turning on, there is a massive short somewhere. You need to find that first, before any concerns about the xStation.

spikeyxx, 2025-03-05:
I'll double check the areas I was working when I get a chance.