x-station/xstation-issues #400: Help me understand why some SD cards load some games and others do not

Tronicgg, 2025-03-11:
Recently bought and installed an Xstation. Very happy overall. I went ahead and bought a 512GB SD card and dumped a, let's say, curated 1G1R set. Every single game I've tried works fine.

Today I went ahead and started adding some english translation patched images. All work fine except for "Racing Lagoon (Japan, Asia)". I tried the image on an emulator and worked fine. So burned a CD and tried on a chipped PS1 and worked fine. I then tried another SD card (as per a reddit user recommendation) and this time it worked fine on the Xstation. 

Why would some games work on some SD cards? At first I thought there must be some config or caché thing going on, so moved over a fresh "00xstation" folder to the SD. No luck. Any ideas? Having a single SD card setup with a more or less 1G1R collection and then having to change cards to play a single game is a little inconvenient.

Tronicgg, 2025-03-11:
Btw, both cards are Amazon basics. One is a 512 and the other a 128.

ramapcsx2, 2025-03-11:
Hey,
you must have been pretty unlucky with your cards. If you want to avoid this in future, I can recommend Sandisk Extreme (red/gold).
They don't particular excel in anything, but are reliably fast enough for every situation in xStation.
Another thing to try is the latest beta firmware on Discord, pinned in the "beta" channel:
https://discord.gg/RRyxdjdH2r
This has significant work done on SD card performance, and might be enough to fix it with your cards.