x-station/xstation-issues #402: font colr/size and gamess taht show no video

ulao, 2025-03-26:
Just gt my xstation up and running. Saw 3 strange things.

1) After the update it reboot I think , to a grey screen. I just power it off and back on to fix this, is that normal?

2) Contra: Legacy of War, using fast boot, boot4ed back to the menu the first time, on the second attempt it booted to the game. After the menu screen a video loaded to a black screen playing the music. It seem to be stuck there. 

3) Can the font be bigger? Or not yellow? I can not really see it too well, it would help to change either or. I did try removing the background but it didn't help.

Thx for all the work for this, it was an easy install, and works amazing!

ramapcsx2, 2025-03-27:
Hey,
the font and such seem to be appreciated by people. This is the first time I hear a request to change the color..
Are you using a Composite Video connection and/or old consumer TV by chance?
None of the background on/off + theme combinations fix it?

The games are expected to work fine, but if there's trouble, please first try a different SD card. Any other card works for testing. You only want to see whether the behaviour changes or not.

ulao, 2025-03-27:
Are you using a Composite Video connection and/or old consumer TV by chance? --YES
None of the background on/off + theme combinations fix it? --NO it didn't 
I can try another SD card.

ulao, 2025-03-28:
looks like its more of a general video out issue with my psx, I see issues in games too. Other consoles do not have the issue.  

issues with the game seems to have gone away.

ulao, 2025-03-29:
guessing these on a pu-8

[image]

ramapcsx2, 2025-03-29:
Ah yea, correct. If you look for more info, you might be able to figure out which are used for your Composite Video line. Otherwise, replace all of them.
Oh, but also: Since you have a PU-8 here, there are 2 outputs that carry CV: The AV plug, and the yellow chinch one. They each have their own capacitor pair, so you can try the other one and see if it's better.