x-station/xstation-issues #403: Xstation not booting, sound glitches on boot

marcosmaiarj, 2025-04-02:
I'll try to be as thorough as possible on the problem i'm having:

I have a japanese SCPH-5501, nice conditions, no problem reading disks. First of all, i swapped the bios chip with the Flash Bios from modzvilleusa. Tested it, it worked just fine.

Then, i installed the xstation kit on my PS1. Apparently, everything went just fine: i checked for continuity, see if there was any accidental pin that was raised... Nothing at all.

When i tried to turn it on, i noticed a sound glitch on the boot (https://youtu.be/7u3gVCH22cQ) and then the memory card screen appeared. I double checked all my soldering and pins, but didn't see anything wrong with it

closed 2025-04-02 by marcosmaiarj

ramapcsx2, 2025-04-02:
Uh yea... And this is solved, I assume? :p

marcosmaiarj, 2025-04-02:
> Uh yea... And this is solved, I assume? :p

Kinda. I rechecked everything and realized i lifted the right pins on the wrong chip (i lifted the spu chip... Dumb mistake) . I'm working on resoldering the SPU chip first and see if i can fix this.