x-station/xstation-issues #404: PS1 Boots to BIOS, not Xstation - Solid Blue Light on XStation + Green Light Blinking Twice

mkbahr, 2025-04-02:
Greetings.

I didn't find any recent issues exactly replicating what I'm experiencing, open or closed, so I thought to create a new issue. 

I wanted to get into soldering and decided that installing an xStation would be a good first project. I followed the PU-18 installation guide on the wiki, and I'm certain I did the steps correctly. My PS1 currently boots to its BIOS and not into the xStation menu. The xStation main board flashes a green light twice on boot, and a solid blue light that stays on until the PS1 is reset or powered off. Interestingly, a config file does seem to be created in the 00xstation folder on the SD card, but it's empty aside from a question-mark-looking character. 

Are pictures of my QSB installation and chip pin-lifting needed?

Thank you.

ramapcsx2, 2025-04-03:
Hey ho,
you also have the very generic case of one or more signals (on the QSB usually) not working. Cold solder joints, accidental shorts, etc.
You can find many tips in the issue tracker here (check closed ones, too. Basically all apply to your case.).
If you want to chat, hop on Discord :)
https://discord.gg/964rCxBB

mkbahr, 2025-04-03:
I posted in the Discord (the xstation-help channel) about starting troubleshooting with checking the BIOS' CD player. From the other issue threads that I've seen, it's not clear to me whether I need to remove the xstation main board before reinstalling the original disc drive, or if the track that the BIOS should try to read is the loader.bin file. Can you give guidance on this?

mkbahr, 2025-04-04:
Then I think I've found the problem. When I go to the CD player in the BIOS, the 0:00 at the top appears for a bit, then disappears, and the play button does nothing (controller works in the BIOS). I assume this means the IC701 is borked?

mkbahr, 2025-04-12:
For the sake of completeness, I will post my solution here. 

I got my multimeter, and testing revealed that all of my solder joints on the QSB were good. Then, I tested the pin pads on IC701, and it revealed that two of the pins pads had been bridged. I removed that bridge, and the next time I booted the PS1, it booted to the xStation menu. 

Now on to the RetroGem. :)

closed 2025-04-12 by mkbahr

ramapcsx2, 2025-04-12:
Glad to hear :)
So as per Discord chat (iirc), the fact the CD Player test would not init the disk at all meant it was likely to be a problem with the SUBQ signals. So I assume the bridged pads were around DSP 78 or so :P