x-station/xstation-issues #405: [Feature Request] Swap Disk  - Choose Disk when lid is open.

TiagoSliachticas, 2025-04-08:
Hello. 

I suggest a new method to Swap Disk (if it be possible). When the Lid is opened a Loader Submenu is opened in the screen with the chosen ISOs at Loader, allowing choose the next ISO to load. 
Or use as scheme like used in PopStarter to swap disk, using controller buttons to order the disks in the custom sequence, not sequencial as is in the present swap disk method.

This sugestion is becouse some games use random disk during the swap, not sequencial, like Monster Rancher

ramapcsx2, 2025-04-09:
It's (almost) impossible to do that. When a game is running, no xStation software runs on the console anymore, and only the low level ODE functions are active.
I can monitor a few signals, such as the lid open / close event, but that's about it.

ramapcsx2, 2025-08-13:
Still the same reply :p
I mean, there is a theoretical way, involving the xStation loader installing patches to all games, which can monitor the pads and then do the complicated .... no, there is no usable way, sorry :p