x-station/xstation-issues #406: Legends of Legaia Soft Crash

edge87, 2025-04-20:
Greetings,

Hardware: SCPH-5501
Fimware: 2.0.2 (Tried Beta 2.2.0 from Discord channel but it doesn't appear to have changed the version number on the menu after install) 

While attempting to play 'Legends of Legaia' the game seems to skip playing a call out and then doesn't continue on. This happens during battle, character will perform a move and normally plays a sound/voice that goes with that move. When it fails to play that sound the background music continues, the camera keeps circling the fight, but the game is soft crashed and will not recover. 

I am confident my installation is properly done as I was able to play another 4 disk RGP from start to finish. 

I have tried 3 different SD Cards: 
 - Gigastone ( U3 /V30 )
 - PNY Elite-X ( U3/V30 ) 
 - SanDisk Extreme ( A2  Red/Gold)

Formatted the SD Cards via the PDF documentation using SD Card Formatter.

ramapcsx2, 2025-04-20:
Hey ho,
you'll need to get off of 2.0.2 for this game. There is a certain bug in it that hits in Legaia. If you were on 2.0.2 before, the update should go through without hassle. You need to update twice if you came from < 2.0.2 only.

edge87, 2025-04-21:
Downgrading to 1.6.1 has solved the problem. Thank you.

closed 2025-04-21 by edge87