x-station/xstation-issues #407: FMV and audio stuttering regularly after 3 months

thunderwave81, 2025-04-25:
Hello,

i'm heaving the issue that FMV and audio are stuttering in almost every game.

The issue first appered 3 month after x-station installation. Up to this point in time the SD-Card wasn't removed from slot or modified (no games deleted or added). At this time i've checked and reformatted the SD-Card without being able to solve the problem.

Finally i solved the problem by installing the 2.2.0 beta (Dec 31 2024) Firmware Update. - Obviously not a SD-card or hardware problem.

The x-station worked fine for the next 3 month and now exactly the same problem appeared again. As before the SD-Card wasn't removed from slot or modified (no games deleted or added), so i can exclude a hardware problem.

I am at a loss as to how I can solve the problem permanently. Will there be a new beta firmware?

ramapcsx2, 2025-04-25:
Hey ho,
the 2.2.0 update helps marginal SD cards quite a lot, though I have an even more substantial tweak in the works.
Either way, the card is going to be the problem. If you can, try out a few other cards, and if you're looking for a proven solid one, the Sandisk Extreme (red/gold) are recommended. Example: https://www.amazon.com/-/de/dp/B082WNC4NK

thunderwave81, 2025-04-25:
Hi,

thank you for the quick reply. I'm currently using the following Sandisk Ultra [](https://www.amazon.de/dp/B0B7NYN3N3/ref=pe_27091401_487024491_TE_item?th=1)
I've made the following performance test (attached Screenshot)
Is this card type really too slow? 

[image]

ramapcsx2, 2025-04-25:
Apparently so, yes. It's quite sad, but their plain Ultra line has become a mixed bag of flash cell quality. 
Testing for this requires a tool that can random seek at the lowest block size, ie 512B. 
The "HDDScan" tool can do this: Butterfly reads, smallest block size, watch for excessively long sectors.
Next run those sectors will be elsewhere on affected cards. It means they need to do too much flash cell maintenance in the background.

thunderwave81, 2025-04-26:
If it is a hardware problem, why does the error not occur from the beginning but only after 3 months? There is no write access to the card, is there?

ramapcsx2, 2025-04-26:
If you check the Discord, I talked about it in length. Note that I'm no expert on this, it's just from what I gathered over time and how I understand it:
The flash cells that store the bits have become tiny. They can loose their charge over time, and the controller firmware in the card manages this by regularly refreshing them (basically a background copy operation). The way this algorithm works differs from brand to brand and between flash cell qualities. Bad cards can really go to town sometimes, to the point of the entire card handle dropping.

ramapcsx2, 2025-04-26:
With this in mind, take the latest super high capacity cards: They must be storing more and more bits in the smallest yet physical flash cells. It's no wonder they are now built with the assumption of permanent refreshes.
This however doesn't fly with the access pattern the xStation uses, which is a stop and go request train of always the next little bit of data (in linear reading, most of the time). Hardware limitations make better strategies hard, but I have one implemented in my devel build atm. This will help the situation somewhat, but it's not ready yet.

thunderwave81, 2025-04-28:
Thank you for the explanations and insights regarding the issue. I've ordered the suggested SD card in hope to solve the problem.

ramapcsx2, 2025-04-28:
I did mention to simply try out any other card (first), right? I hope :p
Sometimes people go right to buying, but you should do any testing you can to verify whether it actually is the card.
Oh well, chances are good here, just from experience :p

thunderwave81, 2025-04-28:
I wasn't aware that there are such huge differences in speed between Sandisk Ultra and Extreme Versions because the pice difference is pretty low. I'll report my further troubleshooting here and in Discord as well. :-)

thunderwave81, 2025-05-03:
The new Sandisk Extreme arrived today. I did a full formatting and a benchmark test after that. Well the speed difference to the old  filled Ultra is not significant, except of Seq writing. I am excited how it works in xstation.

[image]

ramapcsx2, 2025-05-03:
~80MB/s is the top speed for regular card readers. To get more, you need one that supports the vendor specific enhancements.
It used to be that only Sandisk readers (and only new enough) would do this, but these days it seems 3rd parties also have support.
I have an Anker reader here that works, for example.

thunderwave81, 2025-05-10:
New Sandisk Extreme formatted with SD Card Formatter (overwrite format), put games on it, works as expected. Well i'll report status in about 3 month.
Old Sandisk Ultra again formatted with SD Card Formatter (overwrite format), put games on it, works like the new one. But i assume just for 3 month.

ramapcsx2, 2025-05-10:
Nah, they're known to work well. I use them a lot in development here.

thunderwave81, 2025-05-13:
The three-month time period does not seem to be a coincidence. Perhaps most vendors use the same cell refreshing technology.
Did you fill both cards with the same amount of data?

thunderwave81, 2025-05-13:
@Tronicgg You can try to fully reformat and also put same same 3 games on the Amazon as on the Kingston, but i don't think the amount of data is really the problem. My two cards (SanDisk Ultra and SanDisk Extreme) are both filled ~90%.

Tronicgg, 2025-05-13:
@thunderwave81 Just went ahead and ordered a Sandisk Extreme. Let’s see how it goes, I’ll keep an eye on this thread for your experience with the Extreme in the coming months. Currently reformatting the Amazon 512GB again in the hopes it works in the meantime.

I’ve all my consoles ODEed/Everdrived and I have to say, the Xstation is finicky AF when it comes to SD cards. The reason I’ve been using two cards is because English translated roms do not work on the 512GB card for some reason, but they do on the old Kingston card.

ramapcsx2, 2025-05-13:
Maximize your chances:
- don't go for the largest sizes (most dense flash cells), try to get something like 256GB
- the Sandisk red/gold (or red/black) are verified well working
- don't use any SD extender cables, typically found on some 3D print mounts (they have various issues, ask for details)
- format the card to exFAT, using the SD Association tool, pick default settings and quick format
- copy all your games you ever want onto the card in one operation, using one copy job (avoid fragmentation)

That said, you must be pretty unlucky with the cards you got. I test a lot of them, and almost all are just fine.
Though I must admit that I can't really thoroughly test behaviour several months later.. This would have to happen by chance.

Tronicgg, 2025-05-13:
@ramapcsx2 Yeah, I literally did everything on that list when I first set it up 3 months ago with the exception of size and card manufacturer. Anyway, I ordered a Sandisk Extreme earlier today, but went for 512. Having to constantly swap cards is too much of a hassle to go for twin 256. Let’s hope it works well, fingers crossed.

Cheers.