x-station/xstation-issues #408: Problem installation pu-8 late

Arabisnake, 2025-04-30:
Hi
I have successfully installed several xstations, but this last one I have a problem with the apll point on the late pu-8 board. Is there an alternative point?
Thanks!

ramapcsx2, 2025-04-30:
Hey,
if the pad ripped of, please show a picture of what remains.

Arabisnake, 2025-04-30:
I literally don't see any gray pad on the board; it remains green, as if there had never been a pad there. I soldered the cable at the time and the Xstation worked fine, but occasionally when I turned on the console, the console menu would appear. I decided to check the PLL cable, but the pad disappeared.

ramapcsx2, 2025-04-30:
If I don't see the situation, I can't say much :p

Arabisnake, 2025-04-30:
Yes, I understand. After I get home from work, I'll upload a photo.
Thanks bro!

Arabisnake, 2025-05-01:
Finally it wasn't a PLL fault, but rather another point that wasn't making good contact at all. It was quite intermittent and I decided to use the soldering iron again at the point that is close to the IC mechacon, which would be pin 20 conection flex