x-station/xstation-issues #409: Where i can find PU-8 Early diagram?

GTXbox21, 2025-05-04:
Hello! Have qsb for PU-8 Early but i can find diagram only for late PU-8. Where i can find for early? Or is the soldering scheme for early and late PU-8 the same? Desolder the same resistors and lift the same pins on the chip? sorry for my english :)

GTXbox21, 2025-05-04:
Which pins need to be desoldered?
[](https://ibb.co/ymZtmVVj)

ramapcsx2, 2025-05-04:
Oh, an early PU-8 (-12) in an NTSC-J 5000? Curious!
The EARLY 8 guide applies :)

https://github.com/x-station/xstation-releases/wiki/xStation-ODE-installation-guide-for-the-EARLY-PU-8-Playstation-1-motherboard

GTXbox21, 2025-05-05:
> Oh, an early PU-8 (-12) in an NTSC-J 5000? Curious! The EARLY 8 guide applies :)
> 
> https://github.com/x-station/xstation-releases/wiki/xStation-ODE-installation-guide-for-the-EARLY-PU-8-Playstation-1-motherboard

Thank you very much! I tried to find the necessary scheme for about a week everywhere, but I only came across the late pu 8... But all this time the solution was so close...

closed 2025-05-05 by GTXbox21