x-station/xstation-issues #410: Problems Reading Discs, Worth attempting an Xstation?

RaikouSRL, 2025-05-07:
Hi All,

I have PU-7 board here that doesn't read discs with the current drive, the drive spins up really fast once i close the lid and doesn't stop til i power off or open it again, it wont spin up like that when no disc is inserted, i read in a few places it could be caused by the BA6397FP chip or IC305, which i swapped both of them from a donor PU-7 board but i still have the same problem

unfortunately i dont have another disc drive to test, whats the likelyhood an xstation will work in this system? if it wont, then what other potential things could i replace/test in this PU-7 board to get discs working so i can install an xstation? heres a video of my disc issue

https://github.com/user-attachments/assets/7bcf5ef4-28d6-4f95-bb6c-cfda3a595dca

ramapcsx2, 2025-05-08:
Hey ho,
PU-7 in general have rather experimental xStation support. 
I wouldn't recommend buying one for this console, unless you're okay with the chance of failure.
One thing my PU-7 definitely needed was the removal of 3 diodes on the "CX" bus.
If you want to go this route, ping me and I'll find the instructions for you.
Also you need one of the PU-7 QSB, which are perpetually delayed. Not sure if anyone has them by now ><

The drive issue you see is most likely related to low level DSP functions not working right, which xStation would be fine with, but no guarantees :p

RaikouSRL, 2025-05-08:
Excellent, thanks for getting back to me, that'd be awesome if you have those instructions, I'll try my best to source all the parts for this install and give it a try, better than the system going to waste in its current state

> Hey ho, PU-7 in general have rather experimental xStation support. I wouldn't recommend buying one for this console, unless you're okay with the chance of failure. One thing my PU-7 definitely [...]

ramapcsx2, 2025-05-09:
When you have everything, drop by on the Discord help channel :)
https://discord.gg/EbjJpTt4H3