x-station/xstation-issues #413: Games stop working randomly

challenger087, 2025-05-26:
Hey guys,
I’m having an issue with my XStation. I can play games normally, but then it randomly stops working. When I reboot my PS1, it either doesn’t load the XStation and goes straight to the memory card screen, or sometimes it gets stuck on the PlayStation logo.
If I wait around 10 to 15 minutes, it usually starts working again.
Has anyone else experienced this issue?

ramapcsx2, 2025-05-26:
Hey, as mentioned on Discord, please check what the user with the same issue did, just a day or two before :)

challenger087, 2025-05-26:
> Hey, as mentioned on Discord, please check what the user with the same issue did, just a day or two before :)
I didn’t see anyone with the same issue as me in the Discord server. Could you mention me? My profile is "Challenger"

ramapcsx2, 2025-05-28:
Yea, I can't find this now either. It was just funny, since the exact same problem came up the other day. I'll try to find it for you :p

challenger087, 2025-05-28:
Thanks! I ran the test points and everything checked out fine. Then I cleaned the motherboard and the XStation, and the issue disappeared. pretty weird, lol. Anyway, thanks again for your help, and good luck with the mStation release!