x-station/xstation-issues #414: slow boot to bios after board prep

Rubberdug, 2025-05-28:
Decided to check board was still booting ok after doing all prep work before installing qsb, takes about 12 seconds for anything to appear on screen, also no audio, second time Ive had the exact same issue although first time i abandoned install as i didn't realize it was booting at all at the time, re-flowed the whole chip both times to no avail,also on previous board i swapped the chip back out for one with all pins intact, perfectly good laser on another console made little to no effort to spin up,checked all pins/pads next to lifted pins but cant see anything wrong, anyone else encountered/resolved this? pu-18 if that matters, thanks

Rubberdug, 2025-05-29:
Read elsewhere this is normal, i'll carry on with installation

ramapcsx2, 2025-05-29:
Uh.. Alright? :)